rust-embedded/cortex-m-rt #335: region 'FLASH' already defined

dimpolo, 2021-10-19:
I seem to have stumbled accross this issue [region 'FLASH' already defined](https://users.rust-lang.org/t/embedded-rust-lld-error-memory-x-region-flash-already-defined/34788) but the thread is closed by now.
The proposed fix (using the GNU linker) does work for me.

I wonder if I'm doing something wrong or if there's a bug here.
I set up a reproduction here:
https://github.com/dimpolo/linkerbug
The goal would be to have a bootloader sit at the first 32K of FLASH and the main application at the 32K after.

Steps to reproduce:
```shell
git clone https://github.com/dimpolo/linkerbug.git
cd linkerbug/bootloader
cargo build
```

allexoll, 2022-01-20:
having the same issue, switching to GNU-ld fixed it too.

allexoll, 2022-01-20:
if that helps finding where this issue is coming, having a second workspace with self-tests and its own `.cargo/config.toml` that would include link.x again.  Rust-lld accepts it but it fails building, where gnu-ld fails building, alerting that link.x has been included twice.

adamgreig, 2022-01-20:
Are you setting the linker file using `build.rs`, `.cargo/config.toml`, or both? Setting RUSTFLAGS should only take effect once, I thought, so that if you have two competing `.cargo/config.toml` only one of them should win... but perhaps if one side is using `build.rs` and the other also sets RUSTFLAGS it will end up doubly including `link.x`?

allexoll, 2022-01-21:
sorry my previous comment was not very readable, 

The linker file `memory.x`  is created and added in the `build.rs` in a similar fashion as what's done in [stm32l0xx-hal](https://github.com/stm32-rs/stm32l0xx-hal/blob/master/build.rs). 

The structure is as follow:
```
├── .cargo
│   └── config.toml
├── Cargo.toml
├── build.rs
├── examples
│   └── gpio.rs
├── self-tests
│   ├── .cargo
│   │   └── config.toml
│   ├── Cargo.toml
│   ├── README.md
│   └── tests
│       ├── gpio-input-floating.rs
│       ├── ...
│   ├── gpio.rs
│   ├── lib.rs
│   ├── ...
```

The error I had was happening when i was using rust-lld, because both config.toml in `.cargo/` and in `self-tests/.cargo/` would add the `-Tlink.x` option. so building the self test would include `link.x` twice, which would include `memory.x` twice and so `FLASH` region would be already defined the second time. switching to `gnu-ld` would raise a warning saying that `link.x` was included twice, and so i found what was the issue. i just removed the rust flags from `self-tests/.cargo/.config.toml` and switched back to `rust-lld` and that fixed the issue. i'm not sure if there is anything that could have been done there to help prevent that, or to give a more explicit error message.

Hope this is more clear this time!