rust-embedded/cortex-m-rt #340: Using the hard float compiler `thumbv7em-none-eabihf` problematic

robamu, 2021-12-07:
Hi,

I am trying to set up Rust for the first time on a Cortex-M4 based devices

According to the datasheet, the processor supports a single precision FPU:

[image]

But the problem is that I can't get into the `main` function when using the  `thumbv7em-none-eabihf` compiler. It only works when using `thumbv7em-none-eabi`.

Output when flashing is working , binary generated with `thumbv7em-none-eabi`:

```sh
rmueller@power-pinguin:~/Rust/va416xx-workspace(main)$ gdb-multiarch -q -x jlink.gdb target/thumbv7em-none-eabi/debug/examples/blinky
Reading symbols from target/thumbv7em-none-eabi/debug/examples/blinky...
0x00002de6 in ?? ()
Resetting target
Breakpoint 1 at 0x44c: file va416xx-hal/examples/blinky.rs, line 9.
Loading section .vector_table, size 0x400 lma 0x0
Loading section .text, size 0x1a8 lma 0x400
Loading section .rodata, size 0x138 lma 0x5a8
Start address 0x00000400, load size 1760
Transfer rate: 132 KB/sec, 586 bytes/write.

Breakpoint 1, blinky::__cortex_m_rt_main_trampoline () at va416xx-hal/examples/blinky.rs:9
9	#[entry]
(gdb) 
```

Output when flashing does not work with `thumbv7em-none-eabihf`:

```sh
rmueller@power-pinguin:~/Rust/va416xx-workspace(main)$ gdb-multiarch -q -x jlink.gdb target/thumbv7em-none-eabihf/debug/examples/blinky
Reading symbols from target/thumbv7em-none-eabihf/debug/examples/blinky...
__stext () at asm.S:57
57	asm.S: No such file or directory.
Resetting target
Breakpoint 1 at 0x464: file va416xx-hal/examples/blinky.rs, line 9.
Loading section .vector_table, size 0x400 lma 0x0
Loading section .text, size 0x1c0 lma 0x400
Loading section .rodata, size 0x138 lma 0x5c0
Start address 0x00000400, load size 1784
Transfer rate: 134 KB/sec, 594 bytes/write.
```

and nothing is happening. the warning `57	asm.S: No such file or directory.` is also appearing here.
Does anyone have an idea what might be happending here? I'm glad that I can get into the main now, but maybe I can't use the FPU now, I' m not sure about that. What exactly is happending in the `cortex-m-rt` crate when `thumbv7em-none-eabihf` is used?

adamgreig, 2021-12-07:
The main difference when `eabihf` is used is this extra code is run in reset before jumping to user main:

https://github.com/rust-embedded/cortex-m-rt/blob/b145dadc8ea934a10a828e27be0d7079b2c76b20/asm.S#L81-L94

This should normally be fine for a Cortex-M4F core, though. Maybe you could set a breakpoint at the reset vector instead of the user main function, and single-step to check which instruction it's getting stuck on? Alternatively, are you able to interrupt (using ctrl-c) in the broken case and see where the CPU is (try `info reg` to check registers and see what `pc` is, etc)?

robamu, 2021-12-07:
Oh, that looks conspicously similar to some code or sections I have seen in a startup file provided by the vendor:

```c
/*----------------------------------------------------------------------------
  System initialization function
 *----------------------------------------------------------------------------*/
void SystemInit (void)
{

#if defined (__VTOR_PRESENT) && (__VTOR_PRESENT == 1U)
  SCB->VTOR = (uint32_t) &__Vectors;
#endif

#if defined (__FPU_PRESENT) && (__FPU_PRESENT == 1U)
  SCB->CPACR |= ((3U << 10U*2U) |           /* set CP10 Full Access */
                 (3U << 11U*2U)  );         /* set CP11 Full Access */
#endif

#ifdef UNALIGNED_SUPPORT_DISABLE
  SCB->CCR |= SCB_CCR_UNALIGN_TRP_Msk;
#endif
  
  SystemCoreClock = HBO; // default HBO CLK
}
```

 I'll check what you suggest soon

robamu, 2021-12-07:
Hmm this is really odd, I tried to set a breakpoint at `Reset` but something seems to be really wrong:

```sh
rmueller@power-pinguin:~/Rust/va416xx-workspace(main)$ gdb-multiarch -q target/thumbv7em-none-eabihf/debug/examples/blinky
Reading symbols from target/thumbv7em-none-eabihf/debug/examples/blinky...
(gdb) break Reset
Breakpoint 1 at 0x400: file asm.S, line 44.
(gdb) target remote localhost:2331
Remote debugging using localhost:2331
0x00005af4 in ?? ()
(gdb) s
```

adamgreig, 2021-12-07:
Could you try `si` to step a single instruction? You might need to set a later breakpoint as it will take ages to get through the memory initialisation loop, though. I guess the most interesting thing is to see where the CPU is when it crashes/stops progressing, ideally with `info reg` to dump the program counter and register status at that time. It might be stuck initialising memory or it might have hit a fault and gone to the HardFault handler, in which case we could hopefully figure out what caused it.

robamu, 2021-12-07:
Ah, i was already wondering how to do instruction stepping. Not something I need very often :)

```sh
0x00004292 in ?? ()
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x00005af4 in ?? ()
(gdb) si
0x00005abc in ?? ()
(gdb) si
0x00005abe in ?? ()
(gdb) c
Continuing.
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
0x00005af4 in ?? ()
(gdb) 

```

It's weird, everytime I stop it's at 0x5af4.. not sure how to continue from here

adamgreig, 2021-12-07:
Could you share the failing elf file, or maybe run `arm-none-eabi-nm` on it to get function addresses, and see what's at the 0x5af4 address it's stuck at? It's a bit concerning that your stack pointer is 0x1fff_8e28 which is possibly below the bottom of your RAM (assuming it starts at 0x2000_0000), perhaps it's hitting a stack overflow and failing that way. Could you share your `memory.x`?

robamu, 2021-12-07:
There is nothing there.

I've attached both binaries (non-working hard float and working non-hard float)
and the `arm-none-eabi-nm` output.

Ah, the memory file. I initially suspected it as well.

[image]

```txt
MEMORY
{
	FLASH : ORIGIN = 0x00000000, LENGTH = 256K
	/* RAM is a mandatory region. This RAM refers to the SRAM_0 */
	RAM : ORIGIN = 0x1FFF8000, LENGTH = 32K
	SRAM_1 : ORIGIN = 0x20000000, LENGTH = 32K
}
```

[binaries.tar.gz](https://github.com/rust-embedded/cortex-m-rt/files/7672213/binaries.tar.gz)
[blinky_hf_sections.txt](https://github.com/rust-embedded/cortex-m-rt/files/7672215/blinky_hf_sections.txt)
[blinky_sections.txt](https://github.com/rust-embedded/cortex-m-rt/files/7672217/blinky_sections.txt)

robamu, 2021-12-07:
Okay, this is odd: The old program is probably running. The MCU loads 256 kB from an external NVM to the instruction SRAM at boot. There is a blinky program running which also produces some RTT output.

adamgreig, 2021-12-07:
If the MCU loads from an external NVM at boot, does that mean it has a small ROM bootloader that runs initially? If so, is it right for your code to be loaded from address 0 (since the Cortex-M4 core will always start execution from address 0)?

It's very unusual to see SRAM mapped to 0x1FFF_8000, since the Cortex-M4 memory map says the whole region 0x0000_0000-0x1FFF_FFFF is code memory and 0x2000_0000 - 0x3FFF_FFFF is for SRAM, but I guess both regions are Normal memory type and permit execution, so it's OK...

It looks like the `blinky` and `blinky-hf` ELFs in the tarball are identical (and so are the nm outputs) so no further clues there, but indeed it seems like they won't get to address 5af4 anyway. Hard to see how you're ending up at 5af4 when the code you're programming shouldn't ever jump to such an address though. The memory initialisation in cortex-m-rt will zero your BSS and load any mutable statics from flash, so at least in theory it shouldn't matter what's in the rest of the SRAM. Are there any more details or hints from the vendor's startup code?

robamu, 2021-12-08:
The boot process is implemented in hardware if I understood correctly. The datasheet also mentions that all internal SRAMs will be cleared  to 0, but all 256K will always be loaded by default. I have not seen any other details or hints so far

robamu, 2021-12-08:
I just tried stepping through the ASM code for the working binary. There are issues when I do that. If I only set as breakpoint in the main, somehow it managed to get there. Maybe it's just a lucky hit that it's working for that code.. But I'm not sure how to proceed from here

robamu, 2021-12-08:
I now used `asm layout` to get a graphical view of all the instructions.

I now have the same issue on both hard float and non hardfloat code that I can't step to a certain address.

[image]
[image]

robamu, 2021-12-08:
From one problem to another:

```sh
[ARM none EABI] rmueller@power-pinguin:~/Rust/cortex-m-rt(master)$ ./assemble.sh 
+ crate=cortex-m-rt
+ rm -f 'bin/*.a'
+ arm-none-eabi-gcc -g -c -march=armv6s-m asm.S -o bin/cortex-m-rt.o
arm-none-eabi-gcc: fatal error: cannot execute 'cc1': execvp: No such file or directory
compilation terminated.
```

Might have to try a different cross-compiler..

adamgreig, 2021-12-08:
Very odd! What version of `arm-none-eabi-gcc` do you have? `cc1` should have been distributed with it (for example it's in the ARM downloads). Maybe double check you don't have `GCC_EXEC_PREFIX` set in your environment.

robamu, 2021-12-08:
Its a cross-compiler installed with xPacks

```sh
[ARM none EABI] rmueller@power-pinguin:~/Rust/cortex-m-rt(master)$ arm-none-eabi-gcc -v
Using built-in specs.
COLLECT_GCC=arm-none-eabi-gcc
Target: arm-none-eabi
Configured with: /Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/sources/gcc/configure --prefix=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/install/arm-none-eabi-gcc --infodir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/install/arm-none-eabi-gcc/share/doc/info --mandir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/install/arm-none-eabi-gcc/share/doc/man --htmldir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/install/arm-none-eabi-gcc/share/doc/html --pdfdir=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/install/arm-none-eabi-gcc/share/doc/pdf --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --target=arm-none-eabi --with-pkgversion='xPack GNU Arm Embedded GCC x86_64' --enable-languages=c,c++ --disable-mingw-wildcard --enable-plugins --enable-lto --disable-decimal-float --disable-libffi --disable-libgomp --disable-libmudflap --disable-libquadmath --disable-libssp --disable-libstdcxx-pch --disable-nls --disable-threads --disable-tls --with-gnu-as --with-gnu-ld --with-newlib --with-headers=yes --with-python-dir=share/gcc-arm-none-eabi --with-sysroot=/Host/home/ilg/Work/arm-none-eabi-gcc-10.3.1-2.3/linux-x64/install/arm-none-eabi-gcc/arm-none-eabi --with-native-system-header-dir=/include --with-multilib-list=rmprofile --disable-build-format-warnings --with-system-zlib
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.3.1 20210824 (release) (xPack GNU Arm Embedded GCC x86_64) 

```

basically the one with the best Eclipse support. Never had any issues with it but I never tried to compiler an archive from an assembly file before :D

robamu, 2021-12-08:
I managed to build it now with a downloaded cross-compiler.

Removing the .bss and .data initialization leads to odd behaviour where the program just loops at weird points.

If I remove the FPU specific sections, it works of course because the binaries will be identical. I could try to enable the FPU during run-time now, but I still have the feeling that something else is wrong..

adamgreig, 2021-12-08:
Yea, it sure sounds like something else is going wrong and it seems unlikely it's just to do with the FPU initialisation in cortex-m-rt. Could you post the different binaries for eabi/eabihf? The ones you posted earlier are both identical.

It might be worth reading through the startup and linker script in the vendor code just to double check they don't set anything else and do load code to address 0 and so forth. Or perhaps compile a blinky elf with their C toolchain and run `nm` on it to check where the vector table is and see what's loaded into the vector table for stack pointer and reset vector?

robamu, 2021-12-08:
Thats a good idea. I managed to build something with MDK-ARM now and it should be possible to debug the files compiled by MDK-ARM with GCC theoretically as well. I'll report back on my findings..

robamu, 2021-12-08:
MDK-Arm is working without issues like expected. The reset vector does not appear to be at 0x400 though

Here is the disassembly from the debugger

```txt
0x00000330 3275      DCW           0x3275
0x00000332 0000      DCW           0x0000
0x00000334 3279      DCW           0x3279
0x00000336 0000      DCW           0x0000
0x00000338 3389      DCW           0x3389
0x0000033A 0000      DCW           0x0000
0x0000033C 338D      DCW           0x338D
0x0000033E 0000      DCW           0x0000
0x00000340 3455      DCW           0x3455
0x00000342 0000      DCW           0x0000
0x00000344 3459      DCW           0x3459
0x00000346 0000      DCW           0x0000
0x00000348 0383      DCW           0x0383
0x0000034A 0000      DCW           0x0000
0x0000034C 0383      DCW           0x0383
0x0000034E 0000      DCW           0x0000
                 __main:
0x00000350 F8DFD010  LDR.W         sp,[pc,#16]  ; @0x00000364
                 _main_scatterload:
0x00000354 F000FD2A  BL.W          0x00000DAC __scatterload
                 __main_after_scatterload:
0x00000358 4800      LDR           r0,[pc,#0]  ; @0x0000035C
0x0000035A 4700      BX            r0
0x0000035C 5A81      DCW           0x5A81
0x0000035E 0000      DCW           0x0000
                 __rt_lib_shutdown_fini:
0x00000360 F3AF8000  NOP.W         
                 __rt_final_cpp:
0x00000364 89E8      DCW           0x89E8
0x00000366 1FFF      DCW           0x1FFF
   292:                 LDR     R0, =SystemInit 
0x00000368 4806      LDR           r0,[pc,#24]  ; @0x00000384
   293:                 BLX     R0 
0x0000036A 4780      BLX           r0
   294:                 LDR     R0, =__main 
0x0000036C 4806      LDR           r0,[pc,#24]  ; @0x00000388
   295:                 BX      R0 
   296:                 ENDP 
   297:  
   298:  
   299: ; Dummy Exception Handlers (infinite loops which can be modified) 
   300:  
   301: NMI_Handler     PROC 
   302:                 EXPORT  NMI_Handler               [WEAK] 
0x0000036E 4700      BX            r0
   303:                 B       . 
   304:                 ENDP 
   305: HardFault_Handler\ 
   306:                 PROC 
```

The reset vector does not start at 0x400 .. I though the vector table size is fixed for the  given architecture. Maybe this is not the case here?

adamgreig, 2021-12-08:
The CPU always reads a vector table from address 0 at boot, and the vector table always contains a stack pointer word, then a reset vector location word, then a word for the location of each exception and interrupt handler (up to 512 for CM4). But, the reset vector itself can live anywhere in memory - the entry in the vector table is a pointer to it. It's common for the linker to put other functions first, so what comes immediately after the vector table is not necessarily the reset vector itself. And, the amount of space used by the vector table depends on how many interrupts are implemented (or even enabled... if you never use the high-numbered interrupts, you could in principle use those slots for code instead).

From the elf you generate from the C tooling (which presumably blinks the LED OK), can you confirm there is a vector table at address 0, the first word is a stack pointer to 0x2000_0000, and the next word (address 4) points to a reset vector function? From the snippet you posted it's hard to tell for sure what's going on. If that all lines up, perhaps the generated reset vector will have clues as to some device-specific startup that's maybe required.

robamu, 2021-12-08:
Vector table is at address 0, stack pointer points to `0x1fff89e8` . 
I am trying to find out where that number comes from, here is the snipper form the startup file

```asm
; <h> Stack Configuration
;   <o> Stack Size (in Bytes) <0x0-0xFFFFFFFF:8>
; </h>

Stack_Size      EQU     0x00000400

                AREA    STACK, NOINIT, READWRITE, ALIGN=3
Stack_Mem       SPACE   Stack_Size
__initial_sp
```

Reset vector points to `0x369` like expected

adamgreig, 2021-12-08:
It looks like the reset vector just calls `SystemInit` and then calls `main`, so all the startup must take place in `SystemInit`. And seemingly no other special behaviour is required, the stack pointer seems reasonable enough. It runs down, so you expect it to be somewhere above the bottom of the SRAM which is 1fff8000, it's a bit surprising it's only giving you 2536 bytes of stack from a 32kB memory, given there probably isn't anything using the rest of it, but from the startup file perhaps it's deliberately using a small stack?

Does anything look suspicious if you compare the vector table between the working C elf and the not-working Rust elf(s)? Otherwise the only thing left to be suspicious of is the behaviour of `SystemInit` compared to cortex-m-rt's reset vector... or some other spooky problem. It's still not clear to me how you ended up at 0x5af4 or why it gets stuck there though.

robamu, 2021-12-08:
I now managed to run a rust binary generated with the hard float compiler to main.
I tried to replicate the C binary as close as I could by performing the following changes to `asm.S` and `memory.x`:

- Removed `__pre_init` call
- Remove .bss and .data initialization
- Remove barrier calls in the FPU initialization
- Change the stack start to be identical to the one used in the C binary

So `memory.x` looks like this now:

```
/* This is where the call stack will be allocated. */
/* The stack is of the full descending type. */
/* NOTE Do NOT modify `_stack_start` unless you know what you are doing */
/* SRAM_0 can be used for all busses: Instruction, Data and System */
/* SRAM_1 only supports the system bus */
_stack_start = ORIGIN(RAM) + 0x9e8;
```

and the `asm.S`  `Reset` function looks like this

```asm
Reset:
  # ARMv6-M does not initialise LR, but many tools expect it to be 0xFFFF_FFFF
  # when reaching the first call frame, so we set it at startup.
  # ARMv7-M and above initialise LR to 0xFFFF_FFFF at reset.
  ldr r4,=0xffffffff
  mov lr,r4

#ifdef HAS_FPU
  # Conditionally enable the FPU.
  # Address of SCB.CPACR.
  ldr r0, =0xE000ED88
  # Enable access to CP10 and CP11 from both privileged and unprivileged mode.
  ldr r1, =(0b1111 << 20)
  # RMW.
  ldr r2, [r0]
  orr r2, r2, r1
  str r2, [r0]
  # Barrier is required on some processors.
  # dsb
  # isb
#endif

4:
  # Preserve `lr` and emit debuginfo that lets external tools restore it.
  # This fixes unwinding past the `Reset` handler.
  # See https://sourceware.org/binutils/docs/as/CFI-directives.html for an
  # explanation of the directives.
.cfi_def_cfa sp, 0
  push {lr}
.cfi_offset lr, 0

  # Jump to user main function. We use bl for the extended range, but the
  # user main function may not return.
  bl main

  # Trap on return.
  udf

  .cfi_endproc
  .size Reset, . - Reset

```

Of course I'm not happy with these changes and they should not be necessary..

robamu, 2021-12-08:
Okay, I this is the hexdump of the rust binary, generated using `cargo objdump --example blinky -- -O binary blinky.bin` and then `hexdump -C blinky.bin > raw_bin.txt`:

```txt
00000000  00 00 00 20 01 04 00 00  57 04 00 00 7b 05 00 00  |... ....W...{...|
00000010  57 04 00 00 57 04 00 00  57 04 00 00 00 00 00 00  |W...W...W.......|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 57 04 00 00  |............W...|
00000030  57 04 00 00 00 00 00 00  57 04 00 00 57 04 00 00  |W.......W...W...|
00000040  57 04 00 00 57 04 00 00  57 04 00 00 57 04 00 00  |W...W...W...W...|
*
```

The first two vectors do not look correct here. I'm really confused right now

robamu, 2021-12-08:
This might be a flashing related issue. I made a tick at Verify Download in the JLink GDB Server, and the verification failed.

adamgreig, 2021-12-08:
The hexdump looks OK, it has a SP at 0x2000_0000 which is the top of your DataMem0 (since you probably don't have any statics or bss that's likely fine) and a reset vector at 0x00000401 which sounds believable. Are you flashing it the same way as the C elfs? Once it's compiled into an elf they should hopefully be treated exactly the same by flashing tools...

robamu, 2021-12-09:
Alright, I think I found the issue. The ROM protection needs to be switched off by setting a certain bit to one. This was performed in Keil-MDK using a `*.JLinkScript` file. I am still trying to find out how to use a script file like this in the command line, but I now managed to make the hard float image work by sourcing a special gdb file like this with `gdb-multiarch -q --batch -ex 'source prep-flash.gdb'` :

```txt
target remote localhost:2331

echo Disabling watchdog\n
set *0x400210C0 = 0x1ACCE551
set *0x40021008 = 0x0

echo Disabling Instruction Memory protection\n
set *0x40010010 = 0x1
```

But it looks like the problem was completely unrelated in any case..

robamu, 2021-12-10:
Closed for now because the problem was unrelated, everything is working fine now. Nevertheless, thanks a lot for your help :+1:

closed 2021-12-10 by robamu